KaeptnBalu/Arduino_Spot_Welder_V3 #7: Version 3.3

snmcma, 2019-07-24:
Version 3.3 in a working state, or is it only a temporary project? Can I order a fee from Chinese friends?

KaeptnBalu, 2019-07-24:
Version 3.3 is working fine

snmcma, 2019-08-31:
> 
> 
> Version 3.3 is working fine

what could be the problem? in version 3.3.2 the display does not work, in version 3.3.1 the display works fine. Where could there be a mistake?

KaeptnBalu, 2019-09-01:
V3.3.2 reqires a new library versions for the display.
(Adafruit SSD1306 V1.2.9 and Adafruit GFX V1.5.3)
For more details see https://malectrics.eu/2019/06/05/arduino-spot-welder-software-installing-tutorial-v3-3-2/

snmcma, 2019-09-01:
> 
> 
> V3.3.2 reqires a new library versions for the display.
> (Adafruit SSD1306 V1.2.9 and Adafruit GFX V1.5.3)
> For more details see https://malectrics.eu/2019/06/05/arduino-spot-welder-software-installing-tutorial-v3-3-2/

nice! thnks!

snmcma, 2019-11-03:
I used another encoder (which was at hand, I always go all the way to one side (minus) it doesn’t matter which way I turn the encoder. (there is a flaw in the library)
after assembly, my welding worked 1 time and didn’t work anymore (something went wrong). In automatic mode, should there be voltage across the contacts of the electrodes? (I think one of the mosfets struck me, it shows a voltage slightly less than the input)

snmcma, 2019-11-03:
Is it not difficult for you to take large photographs from all sides? I want to make sure the assembly is correct.

KaeptnBalu, 2019-11-03:
Hi, 
Did you take a look at the assembly instructions: https://malectrics.eu/pcb-set-v3-3-assembly-instructions/
There are a lot of pictures, so you can check if you got everything correct.

Measuring the battery voltage at the welding tips is normal. If it instantly sparks when you touch the nickel, then there is a problem somewhere.

snmcma, 2019-11-03:
[image] 
he writes the data of the current pulse, but there was no pulse on the electrodes. after each pulse, you need to click on the encoder so that I can try again to make a pulse, is this normal behavior?

snmcma, 2019-11-03:
This is not too much current I had on the pulse, mosfets could burn out? I used a powerful 160,000 microfarad capacitor in parallel with the power.

snmcma, 2020-05-17:
Can you give an answer what is the reason? It works on a small power supply, but when I connect it to a car battery, I don’t learn to weld. firmware updated to 3.3.4