KaeptnBalu/Arduino_Spot_Welder_V3 #11: V3.3.3 Arduino Code selectedSubMenu not assigned value correctly

richard-scott, 2020-03-15:
I've migrated your code over to using PlatformIO, and I get this error when compiling it:

> Arduino Spot Welder/src/Arduino_Spot_Welder_V3_3_3.ino:539:37: warning: statement has no effect [-Wunused-value]
>                      selectedSubMenu == 0;

It is from this section:
[image]

My repo is here:
https://github.com/richard-scott/Arduino_Spot_Welder_V3

richard-scott, 2020-04-01:
I've squashed the error in my repo here:
https://github.com/richard-scott/Arduino_Spot_Welder_V3
I'll create a PR.

closed 2020-04-01 by richard-scott

KaeptnBalu, 2020-04-01:
Awesome Job Richard. Thank you so much for your work. This actually fixed the 10.000 welds counter bug. It can now count above 10.000 without creating an error. I will implement your fix in a new V3.3.4 of the spot welder software.