KaeptnBalu/Arduino_Spot_Welder_V3 #17: Weld current estimate to check for propper weld

geo99918, 2020-11-04:
I saw that the weld current is estimated based on the source voltage during the weld pulse, why not use this estimation as a check for a proper weld and trigger an alarm when the current was too low, or even add a menu to set the min and max values for weld current.
E.g. with a discharged car battery 12V is present at idle but during the weld pulse the voltage drops to less than 3 volts and a good weld is not created.
A more advanced version of this would be to measure the total energy delivered to the weld but this might be for another day.

Regards,
George

KaeptnBalu, 2020-11-05:
A llittle problem with the weld current measurement is that it is not super accurate. The welder uses a fixed resistance value to calculate the current flow. It can not measure the resistance itself, just the voltage drop during the pulse. This resistance can change quite a bit depending how the welding tips are placed on the nickel and how good the contact of the wleding cables is. 

I think the idea to set a weld current alarm is pretty good. For that the current measurement should be accurate enough.

To calculate the total energy during pulse and then base the pulse length on that, the current measurement needs to be more precise.  If you got an idea how the welder could do a precise resistance measurement (without complicated hardware) before a weld, that would be great. I think an option to choose between energy based welding and fixed time welding would be preatty cool.

geo99918, 2020-11-05:
Maybe the current can be measured by measuring the voltage drop across a big shunt.
if current and voltage are known during the weld event a better estimation can be done if the weld was successful or not.
Maybe the resistance can be estimated using V and I but the power source internal resistance can't be ignored in this calculation.
Regarding the alarm, a buzzer would be nice to indicate that something went wrong, usually when welding the eyes are on the weld not on the screen.

Satys27, 2025-02-05:
Ok wait for my Project release, I have improved lots of features like robotic arm will weld you just need to seat tight infront of it.